hello-robot/stretch_body #44: dynamixel state reader crashes if the thread is starved

soumith, 2021-06-16:
Here's a minimal reproducible example that crashes (after ~5 minutes or so):

```
#!/usr/bin/env python
import time
import math
import pyrealsense2 as rs
import stretch_body.robot
from stretch_body.hello_utils import ThreadServiceExit

robot=stretch_body.robot.Robot()
robot.startup()

try:
    while True:
        for x in range(100 * 100):
            for x2 in range(100 * 100):
                x = math.log(float(1000.125))
        print("booooooooooooooooooooooooooooooooooooo", x)
        # time.sleep(0.1)
except (KeyboardInterrupt, SystemExit, ThreadServiceExit) as e:
    pass

robot.stop()
```

This line in the dynamixel SDK crashes: https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/b19434d49b353d00b99b82c1b52d52c99fa36fab/python/src/dynamixel_sdk/group_sync_read.py#L139

This is the stack-trace:

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/robot.py", line 41, in run
    self.robot._pull_status_dynamixel()
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/robot.py", line 324, in _pull_status_dynamixel
    self.head.pull_status()
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/dynamixel_X_chain.py", line 82, in pull_status
    pos = self.sync_read(self.readers['pos'])
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/dynamixel_X_chain.py", line 155, in sync_read
    values = [get_val(self.motors[mk].motor.dxl_id) for mk in self.motors.keys()]
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/dynamixel_X_chain.py", line 155, in <listcomp>
    values = [get_val(self.motors[mk].motor.dxl_id) for mk in self.motors.keys()]
  File "/home/hello-robot/droidlet/droidlet/lowlevel/hello_robot/remote/stretch_body/body/stretch_body/dynamixel_X_chain.py", line 135, in get_val
    b = reader.getData(id_num, reader.start_address, reader.data_length)
  File "/opt/ros/noetic/lib/python3/dist-packages/dynamixel_sdk/group_sync_read.py", line 145, in getData
    self.data_dict[dxl_id][address - self.start_address + 3]))
IndexError: list index out of range
```

A few frames before the crash, one sees this in stderr `[ERROR] [head]: Pull Status IOError on: /dev/hello-dynamixel-head`.
The full stack-trace is below.

Adding a few debug prints in the dynamixel SDK directly, one sees this behavior:

```
# A regular / normal set of variable values
self.data_dict: [247, 21, 0, 0]
address: 132
self.start_address: 132
self.data_length: 4
data_length: 4

# Crashing variable values
self.data_dict: [0, 89, 17]
address: 132
self.start_address: 132
self.data_length: 4
data_length: 4
```

So, clearly self.data_dict is supposed to be of length 4, but seems to only be of length 3.

soumith, 2021-06-16:
I suspect some kind of thread-starvation and that you don't handle the error codes returned from Dynamixel correctly.
I can take a closer look tomorrow, but if anyone has ideas, do post what you know.

soumith, 2021-06-16:
here's a much quicker repro of the issue.
the dynamixel thread crashes right away:

```python
#!/usr/bin/env python
import time
import math
import numpy as np
import stretch_body.robot
from stretch_body.hello_utils import ThreadServiceExit

robot=stretch_body.robot.Robot()
robot.startup()

try:
    while True:
        x = np.random.rand(3, 1000, 1000)
        x.tolist()
except (KeyboardInterrupt, SystemExit, ThreadServiceExit) as e:
    pass

robot.stop()
```

aedsinger, 2021-06-22:
Thanks for the condensed test case. We are able to reproduce this and are working on it.

aedsinger, 2021-06-24:
We've a PR here to fix this issue:

https://github.com/hello-robot/stretch_body/pull/49

For background:

### Explanation of DXL Threading issues

When an IO Bound and CPU Bound thread compete for time within a Python process, the IO Bound thread can get starved. [This has a good explanation](https://cloudsek.com/how-do-you-achieve-concurrency-with-python-threads/) of the phenomena. 

In this Git Issue, the test case creates IOErrors given a CPU bound thread.

When doing a packet transaction, the Dynamixel Packet Handler sets a timeout for that transaction based on the size of the packet. 

self.packet_timeout = (self.tx_time_per_byte * packet_length) + (LATENCY_TIMER * 2.0) + 2.0

If the transaction hasn’t completed within this timeout then an IOError gets generated. This can happen when:

1. Start of new transaction. Mark t=0
2. Do a polling serial.read: GIL is yielded to the CPU bound thread
3. CPU bound thread starves the IO bound thread of the GIL for a period
4. Return to the polling serial read. The timeout period has been exceeded and an IO Error is generated

In the test case we are seeing transaction times of over 100ms on occasion. Typical times without a high CPU load are around 15-20ms.

By default the Robotis SDK has:

#define LATENCY_TIMER 16 (ms)

[Dynamixel recommends](http://docs.ros.org/en/kinetic/api/dynamixel_sdk/html/port__handler__linux_8cpp_source.html) relaxing this value when using GroupSyncRead as well. The PR adds the ability to set the latency timer via YAML, and sets it to 64 by default.

DLu, 2021-07-07:
Now that #49 is merged, should this get closed?

closed 2021-07-07 by soumith