hello-robot/stretch_body #143: Initiating Device Class object with default init exits the program.

hello-fazil, 2022-07-26:
The Error is  thrown while initializing initializing Device() class object with no arguments given  (Default :```__init__(name='',req_paramreq_params=True)```)

```bash
In [2]: import stretch_body.device

In [3]: log_device=stretch_body.device.Device()
[WARNING] [root]: Parameters for device  not found. Check parameter YAML and device name. Exiting...
An exception has occurred, use %tb to see the full traceback.

SystemExit: 1

/home/hello-robot/.local/lib/python3.8/site-packages/IPython/core/interactiveshell.py:3406: UserWarning: To exit: use 'exit', 'quit', or Ctrl-D.
  warn("To exit: use 'exit', 'quit', or Ctrl-D.", stacklevel=1)

In [4]: %tb
---------------------------------------------------------------------------
SystemExit                                Traceback (most recent call last)
Input In [3], in <cell line: 1>()
----> 1 log_device=stretch_body.device.Device()

File ~/.local/lib/python3.8/site-packages/stretch_body/device.py:41, in Device.__init__(self, name, req_params)
     39 if self.params == {} and req_params:
     40         self.logger.warning('Parameters for device %s not found. Check parameter YAML and device name. Exiting...' % self.name)
---> 41         sys.exit(1)
     43 self.timestamp = DeviceTimestamp()
     44 self.thread_active = False

SystemExit: 1
```

It looks like the snippet below from [Line 39](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/device.py#L39) is causing the issue.
```
        if self.params == {} and req_params:
                self.logger.warning('Parameters for device %s not found. Check parameter YAML and device name. Exiting...' % self.name)
                sys.exit(1)
```

hello-fazil, 2022-07-27:
@aedsinger Should the Default value of the init argument `req_params`of [Device()](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/device.py#L33) class defined in [Line 33](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/device.py#L33) be set to False `req_params=False` ?

aedsinger, 2022-07-27:
I think the correct fix is to pass use_params=False on line 9 of
firmware_updater.py

[image: image.png]

closed 2022-07-28 by hello-fazil