hello-robot/stretch_body #156: wait_until_at_setpoint() not working

hello-binit, 2022-09-22:
Running the following script on Ubuntu 20.04, Stretch Body 0.4.8:

```py3
import stretch_body.robot
import sys

r = stretch_body.robot.Robot()
r.startup()

if not r.is_calibrated():
    print('home the robot first')
    sys.exit(1)
r.stow()

def move_all():
    r.lift.move_to(0.5137053394519911)
    r.push_command()
    r.end_of_arm.get_joint('wrist_roll').move_to(-0.7439806821245359)
    r.end_of_arm.get_joint('wrist_yaw').move_to(-0.7676295526044397)
    r.end_of_arm.get_joint('wrist_pitch').move_to(-0.7761942786701345)
    r.head.get_joint('head_pan').move_to(-0.8244375296703885)
    r.head.get_joint('head_tilt').move_to(-0.0010067811717669514)

move_all()
r.lift.wait_until_at_setpoint()

r.stop()
```

We expect the robot to move the pose commanded in `move_all()` before exiting the program, but it exits early and the pose is not achieved. Here's the associated log:

```
$ cat ~/stretch_user/log/stretchbody_20220922144628.log
[DEBUG   ] [09/22/2022 14:46:28.693] [pimu]: Starting TransportConnection on: /dev/hello-pimu
[DEBUG   ] [09/22/2022 14:46:28.693] [hello-motor-left-wheel]: Starting TransportConnection on: /dev/hello-motor-left-wheel
[DEBUG   ] [09/22/2022 14:46:28.695] [hello-motor-right-wheel]: Starting TransportConnection on: /dev/hello-motor-right-wheel
[DEBUG   ] [09/22/2022 14:46:28.696] [hello-motor-lift]: Starting TransportConnection on: /dev/hello-motor-lift
[DEBUG   ] [09/22/2022 14:46:28.699] [hello-motor-arm]: Starting TransportConnection on: /dev/hello-motor-arm
[DEBUG   ] [09/22/2022 14:46:28.709] [wacc]: Starting TransportConnection on: /dev/hello-wacc
[DEBUG   ] [09/22/2022 14:46:28.719] [robot]: Starting up Robot stretch-re1-1060 of batch Kendrick
[DEBUG   ] [09/22/2022 14:46:28.819] [head_pan]: [Dynamixel ID:011] ping Succeeded. Dynamixel model number : 1080. Baud 115200
[DEBUG   ] [09/22/2022 14:46:28.874] [head_tilt]: [Dynamixel ID:012] ping Succeeded. Dynamixel model number : 1060. Baud 115200
[WARNING ] [09/22/2022 14:46:28.919] [head_tilt]: Dynamixel head_tilt: Servo is in single-turn mode yet has invalid homing offset of -1682. 
 This may cause unexpected results if not set to zero (using REx_dynamixel_jog.py)
 Please contact Hello Robot support for more information
[DEBUG   ] [09/22/2022 14:46:28.961] [stretch_gripper]: [Dynamixel ID:014] ping Succeeded. Dynamixel model number : 1060. Baud 115200
[DEBUG   ] [09/22/2022 14:46:29.018] [wrist_pitch]: [Dynamixel ID:015] ping Succeeded. Dynamixel model number : 1120. Baud 115200
[DEBUG   ] [09/22/2022 14:46:29.087] [wrist_roll]: [Dynamixel ID:016] ping Succeeded. Dynamixel model number : 1020. Baud 115200
[DEBUG   ] [09/22/2022 14:46:29.153] [wrist_yaw]: [Dynamixel ID:013] ping Succeeded. Dynamixel model number : 1080. Baud 115200
[DEBUG   ] [09/22/2022 14:46:29.291] [base]: Fast motion turned off
[DEBUG   ] [09/22/2022 14:46:29.381] [NonDXLStatusThread]: Missed target loop rate of 25.00 Hz for NonDXLStatusThread. Currently 23.56 Hz
[DEBUG   ] [09/22/2022 14:46:30.372] [tool_stretch_dex_wrist]: Dynamixel X sync read txRxPacket failed with error code = -3001
[DEBUG   ] [09/22/2022 14:46:30.386] [DXLStatusThread]: Missed target loop rate of 15.00 Hz for DXLStatusThread. Currently 5.56 Hz
[DEBUG   ] [09/22/2022 14:46:30.552] [base]: Fast motion turned on
[DEBUG   ] [09/22/2022 14:46:31.119] [robot]: ---- Shutting down robot ----
[DEBUG   ] [09/22/2022 14:46:31.135] [robot]: Shutting down NonDXLStatusThread
[DEBUG   ] [09/22/2022 14:46:31.158] [robot]: Shutting down DXLStatusThread
[DEBUG   ] [09/22/2022 14:46:31.158] [robot]: Shutting down pimu
[DEBUG   ] [09/22/2022 14:46:31.261] [pimu]: Shutting down TransportConnection on: /dev/hello-pimu
[DEBUG   ] [09/22/2022 14:46:31.297] [robot]: Shutting down base
[DEBUG   ] [09/22/2022 14:46:31.302] [hello-motor-left-wheel]: Shutting down Stepper on: /dev/hello-motor-left-wheel
[DEBUG   ] [09/22/2022 14:46:31.406] [hello-motor-left-wheel]: Shutting down TransportConnection on: /dev/hello-motor-left-wheel
[DEBUG   ] [09/22/2022 14:46:31.442] [hello-motor-right-wheel]: Shutting down Stepper on: /dev/hello-motor-right-wheel
[DEBUG   ] [09/22/2022 14:46:31.546] [hello-motor-right-wheel]: Shutting down TransportConnection on: /dev/hello-motor-right-wheel
[DEBUG   ] [09/22/2022 14:46:31.582] [robot]: Shutting down lift
[DEBUG   ] [09/22/2022 14:46:31.585] [hello-motor-lift]: Shutting down Stepper on: /dev/hello-motor-lift
[DEBUG   ] [09/22/2022 14:46:31.691] [hello-motor-lift]: Shutting down TransportConnection on: /dev/hello-motor-lift
[DEBUG   ] [09/22/2022 14:46:31.728] [robot]: Shutting down arm
[DEBUG   ] [09/22/2022 14:46:31.731] [hello-motor-arm]: Shutting down Stepper on: /dev/hello-motor-arm
[DEBUG   ] [09/22/2022 14:46:31.837] [hello-motor-arm]: Shutting down TransportConnection on: /dev/hello-motor-arm
[DEBUG   ] [09/22/2022 14:46:31.876] [robot]: Shutting down head
[DEBUG   ] [09/22/2022 14:46:31.884] [robot]: Shutting down wacc
[DEBUG   ] [09/22/2022 14:46:31.984] [wacc]: Shutting down TransportConnection on: /dev/hello-wacc
[DEBUG   ] [09/22/2022 14:46:32.021] [robot]: Shutting down end_of_arm
[DEBUG   ] [09/22/2022 14:46:32.031] [robot]: ---- Shutdown complete ----
```

aedsinger, 2022-09-23:
What is the value from:  stretch_params.py | grep lift | grep pos_near

aedsinger, 2022-09-23:
Also what version of robot is this (RE1 or RE2)?

aedsinger, 2022-09-23:
Also, can you post the output of r.lift.pretty_print() just before the call to r.stop()?

hello-binit, 2022-09-23:
I'm on an RE1 from the Kendrick batch. Here's the `pos_near_setpoint_d` param:

```
$ stretch_params.py | grep lift | grep pos_near
stretch_body.robot_params.nominal_params                               param.hello-motor-lift.gains.pos_near_setpoint_d                       4.0
```

Here's what pretty_print gives me when running [this script](https://gist.github.com/a45cbc144f3a38ea76fb9231070f9853):
```
$ python3 wai.py 
[WARNING] [head_tilt]: Dynamixel head_tilt: Servo is in single-turn mode yet has invalid homing offset of -1682. 
 This may cause unexpected results if not set to zero (using REx_dynamixel_jog.py)
 Please contact Hello Robot support for more information
--------- Stowing Arm ----
--------- Stowing ToolStretchDexWrist ----
--------- Stowing Lift ----
----- Lift ------ 
Pos (m):  0.19995437079493245
Vel (m/s):  -0.000391568875423013
Soft motion limits (m) [0.12, 1.0972310016334053]
Timestamp PC (s): 1663960894.4014077
-----------
Mode MODE_POS_TRAJ
x_des (rad) 53.79509735107421 (deg) 3082.232036711947
v_des (rad) 9.948376736367681 (deg) 570.0000000000001
a_des (rad) 15.707963267948966 (deg) 900.0
Stiffness 1.0
Feedforward 0.75
Pos (rad) 20.932538986206055 (deg) 1199.3461384026618
Vel (rad/s) -0.1687108874320984 (deg) -9.66642180776596
Effort (Ticks) 41.015380859375
Effort (Pct) 16.587102553423712
Current (A) 0.5307872817095588
Error (deg) 0.6337890478979388
Debug 0.0
Guarded Events: 6
Diag 00000000000000000100100100010101
       Position Calibrated: True
       Runstop on: False
       Near Pos Setpoint: True
       Near Vel Setpoint: False
       Is Moving: True
       Is Moving Filtered: False
       At Current Limit: False
       Is MG Accelerating: False
       Is MG Moving: False
       Encoder Calibration in Flash: True
       In Guarded Event: False
       In Safety Event: False
       Waiting on Sync: True
Waypoint Trajectory
       State: idle
       Setpoint: (rad) 0.0 | (deg) 0.0
       Segment ID: 0
Timestamp (s) 1663960894.381975
Read error 0
Board variant: Stepper.Kendrick.V1
Firmware version: Stepper.v0.1.0p1
```

aedsinger, 2022-09-23:
I believe the issue is that you need a time.sleep() after the push command in order to wait for motion to commence. If this resolves the issue then we can explore a way to avoid the need for the sleep in the Stretch Body design

hello-binit, 2024-02-01:
Fixed with the addition of `wait_while_is_moving()`:

https://github.com/hello-robot/stretch_body/blob/289fcaebf31579ff399828c1b24b567bba65e40f/body/stretch_body/stepper.py#L465-L476

closed 2024-02-01 by hello-binit