hello-robot/stretch_body #174: Using dry_erase_holder_v1 with the stretch body tools

xiangzhi, 2023-03-02:
Hi,

I attached the tool `dry_erase_holder_v1` onto our robot, updated the tool parameters in the `stretch_user_params.yaml` according the [instructions here](https://docs.hello-robot.com/0.2/stretch-tutorials/stretch_body/tutorial_tool_change/#toolnone), and was trying to use the given tools such as `stretch_robot_home.py` and `stretch_xbox_controller_teleop.py`. Both kept giving me error about trying to home the `wrist_yaw`. Is this expected behavior? 

Thank you!

hello-fazil, 2023-03-13:
Hi, @xiangzhi thank you for reporting this. We are working on reproducing this issue and will get back to you soon with a fix. I appreciate your patience.

hello-fazil, 2023-03-13:
Hi @xiangzhi, We found the wrong baud rate set for the dry_erase_holder_v1 tool's parameters. This bug is [fixed](https://github.com/hello-robot/stretch_tool_share/commit/8f4b49db50abaaff6561a98ad634513321569629) now.
You can update the Stretch Tool Share library with the latest:
```
pip3 install -U hello-robot-stretch-tool-share
```
Let us know if the issue still persists.

closed 2023-03-13 by hello-fazil

xiangzhi, 2023-03-14:
Awesome. Thank you for the fix.