hello-robot/stretch_body #185: dynamixel_hello_XL430 pull_status error from data in internal thread

youliangtan, 2023-05-18:
Occasionally Iam getting error when using the stretch_body sdk. The error below are multiple instances when I received the error. It seems to be some mishandling of the `data` variable in a separate thread, which can be `None` in some race access scenario.

The below printouts are just a snapshot of the error when it happens.

```bash
{'ts': 1684440402.6173935, 'x': 6226, 'v': 0, 'eff': 0, 'temp': 39, 'err': 0}
Received request: b'{"compact_status": false}'
Received request: b'{"move": {"yaw": -0.05}, "compact_status": true}'
{'ts': 1684440403.015795, 'x': 6226, 'v': 0, 'eff': 0, 'temp': 39.0, 'err': 0}
Lift force 0
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 40, in run
    self.robot._pull_status_dynamixel()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 380, in _pull_status_dynamixel
    self.head.pull_status()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_X_chain.py", line 200, in pull_status
    self.motors[mk].pull_status(data)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 316, in pull_status
    self.status['pos'] = self.ticks_to_world_rad(float(x))
TypeError: float() argument must be a string or a number, not 'NoneType'

########################################################################################

Received request: b'{"compact_status": false}'
Received request: b'{"compact_status": false}'
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 40, in run
    self.robot._pull_status_dynamixel()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 379, in _pull_status_dynamixel
    self.end_of_arm.pull_status()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_X_chain.py", line 200, in pull_status
    self.motors[mk].pull_status(data)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/stretch_gripper.py", line 60, in pull_status
    DynamixelHelloXL430.pull_status(self,data)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 316, in pull_status
    self.status['pos'] = self.ticks_to_world_rad(float(x))
TypeError: float() argument must be a string or a number, not 'NoneType'

########################################################################################

Received request: b'{"compact_status": false}'
{'ts': 1684440794.308914, 'x': 6226, 'v': 0, 'eff': 0, 'temp': 39.0, 'err': 0}
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 40, in run
    self.robot._pull_status_dynamixel()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 380, in _pull_status_dynamixel
    self.head.pull_status()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_X_chain.py", line 200, in pull_status
    self.motors[mk].pull_status(data)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 316, in pull_status
    self.status['pos'] = self.ticks_to_world_rad(float(x))
TypeError: float() argument must be a string or a number, not 'NoneType'
Received request: b'{"compact_status": false}'
Received request: b'{"move": {"y": 0.1, "z": 0.75, "roll": 0, "pitch": -0.3, "yaw": 0, "gripper": 55}, "compact_status": true}'
Lift force 0
[INFO] [robot_monitor]: Base bump event
```

I definitely can look into details of how the race condition actually happens, but for the quick fix, i just apply a quick "hack" to [this block](https://github.com/hello-robot/stretch_body/blob/1d887bbe5afbcd63d082e427dbb76b577144d6b5/body/stretch_body/dynamixel_hello_XL430.py#L314-L326):

```python
        #Now update status dictionary
        if pos_valid and x is not None:
            self.status['pos_ticks'] = x
            self.status['pos'] = self.ticks_to_world_rad(float(x))
        if vel_valid and v is not None:
            self.status['vel_ticks'] = v
            self.status['vel'] = self.ticks_to_world_rad_per_sec(float(v))
        if eff_valid and eff is not None:
            self.status['effort_ticks'] = eff
            self.status['effort'] = self.ticks_to_pct_load(float(eff))
        if temp_valid and temp is not None:
            self.status['temp'] = float(temp)
        if err_valid:
            self.status['hardware_error'] = err
```

However, I am afraid that some problems might arise due to this hack.

fyi: iam using [this repo](https://github.com/Healthcare-Robotics/stretch_remote/) to test out a brand new stretch v2.

hello-binit, 2023-05-19:
Hi @youliangtan, thanks for reporting this and sharing your project, [stretch_remote](https://github.com/Healthcare-Robotics/stretch_remote/) looks really cool! I hope I can try it out!

I don't think there's anything wrong with your quick fix, you may just have stale data for a few loops. To really debug what's going on, we will need to add debugger checkpoints and/or print statements in the [`dynamixel_X_chain.DynamixelXChain.pull_status()`](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/dynamixel_X_chain.py#L169-L173) method. Specifically, the errors you saw should have resulted in a "Dynamixel communication error during pull_status on \<device name\>" warning to tell us which Dxl device caused the issue, as well logging the issue to [`DynamixelCommErrorStats`](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/dynamixel_hello_XL430.py#L42-L48) so we could determine the frequency at which the dxl fails to deliver its position data.

hello-binit, 2024-02-01:
Fixed in https://github.com/hello-robot/stretch_body/pull/222/commits/04c4a676d02f7ee83f531d28734ee7ce70ba36c6

closed 2024-02-01 by hello-binit