hello-robot/stretch_body #247: speed control using xbox controller

Minyoung1005, 2024-01-24:
Hi, I am testing the gamepad_teleop.py and get some errors:

When I first turn on the stretch and run the code, calibration runs well but the code suddenly stops with this error: 
`hello-robot@stretch-re2-2003:~/stretch_body_minyoung$ python body/stretch_body/gamepad_teleop.py 
Key mapped to End-Of-Arm Tool: tool_stretch_dex_wrist
[WARNING] [robot]: Not able to use asyncio for transport communications. Defaulting to sync.
press the start button to calibrate the robot
press the start button to calibrate the robot
press the start button to calibrate the robot
press the start button to calibrate the robot
press the start button to calibrate the robot
press the start button to calibrate the robot
--------- Homing Head ----
--------- Homing Lift ----
Homing Lift...
Hardstop detected at motor position (rad) 90.21973419189453
Marking Lift position to 1.098445 (m)
Marking Lift position to 0.000000 (m)
[INFO] [robot_monitor]: Guarded contact lift
[INFO] [robot_monitor]: Base bump event
Lift homing successful
--------- Homing Arm ----
Homing Arm...
Hardstop detected at motor position (rad) -7.27034330368042
Marking Arm position to 0.000000 (m)
[INFO] [robot_monitor]: Guarded contact arm
[INFO] [robot_monitor]: Wrist single tap: 6
[INFO] [robot_monitor]: Wrist single tap: 8
[INFO] [robot_monitor]: Wrist single tap: 9
[INFO] [robot_monitor]: Wrist single tap: 10
[INFO] [robot_monitor]: Wrist single tap: 11
[INFO] [robot_monitor]: Wrist single tap: 12
Arm homing successful
Moving to first hardstop...
First hardstop contact at position (ticks): 90
-----
Homing offset was 2471
Marking current position to zero ticks
Homing offset is now  2376 (ticks)
-----
Current position (ticks): 46
Moving to calibrated zero: (rad)
Moving to first hardstop...
First hardstop contact at position (ticks): -8514
-----
Homing offset was -2287
Marking current position to zero ticks
Homing offset is now  6222 (ticks)
-----
Current position (ticks): 32
Moving to calibrated zero: (rad)
[INFO] [robot_monitor]: Wrist single tap: 14
Exception in thread SystemMonitorThread:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 167, in run
    self.step()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 155, in step
    self.robot._step_sentry()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 560, in _step_sentry
    self.end_of_arm.step_sentry(self)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_X_chain.py", line 267, in step_sentry
    self.motors[k].step_sentry(robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_tool_share/stretch_dex_wrist/wrist_pitch.py", line 34, in step_sentry
    DynamixelHelloXL430.step_sentry(self,robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 459, in step_sentry
    self._update_safety_vel_brake_zone()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 468, in _update_safety_vel_brake_zone
    brake_zone_factor = self.params['motion']['vel_brakezone_factor'] # Propotional value, for now value 1 seems to work fine with all Dxl joints
KeyError: 'vel_brakezone_factor'
Traceback (most recent call last):
  File "body/stretch_body/gamepad_teleop.py", line 429, in <module>
    gamepad_teleop.mainloop()
  File "body/stretch_body/gamepad_teleop.py", line 399, in mainloop
    self.do_motion()
  File "body/stretch_body/gamepad_teleop.py", line 227, in do_motion
    self.command_robot_joints(robot)
  File "body/stretch_body/gamepad_teleop.py", line 154, in command_robot_joints
    self.wrist_pitch_command.stop_motion(robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/gamepad_joints.py", line 490, in stop_motion
    motor.set_velocity(0,self.params['motion']['max']['accel'])
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 551, in set_velocity
    if self.params['set_safe_velocity'] and self.in_vel_brake_zone: # only when sentry is active
KeyError: 'set_safe_velocity'`

After I try to run the same code, I only get the following error, which might be just because the controller is already connected to the robot.

`hello-robot@stretch-re2-2003:~/stretch_body_minyoung$ python body/stretch_body/gamepad_teleop.py 
Key mapped to End-Of-Arm Tool: tool_stretch_dex_wrist
[WARNING] [wrist_yaw]: WARNING: Servo wrist_yaw in error state: overload_error. Reboot servo with stretch_robot_dynamixel_reboot.py
[WARNING] [robot]: Not able to use asyncio for transport communications. Defaulting to sync.
[WARNING] [robot_monitor]: Dynamixel overload_error on tool_stretch_dex_wrist:wrist_yaw
Exception in thread SystemMonitorThread:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 167, in run
    self.step()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 155, in step
    self.robot._step_sentry()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/robot.py", line 560, in _step_sentry
    self.end_of_arm.step_sentry(self)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_X_chain.py", line 267, in step_sentry
    self.motors[k].step_sentry(robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_tool_share/stretch_dex_wrist/wrist_pitch.py", line 34, in step_sentry
    DynamixelHelloXL430.step_sentry(self,robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 459, in step_sentry
    self._update_safety_vel_brake_zone()
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 468, in _update_safety_vel_brake_zone
    brake_zone_factor = self.params['motion']['vel_brakezone_factor'] # Propotional value, for now value 1 seems to work fine with all Dxl joints
KeyError: 'vel_brakezone_factor'
Traceback (most recent call last):
  File "body/stretch_body/gamepad_teleop.py", line 429, in <module>
    gamepad_teleop.mainloop()
  File "body/stretch_body/gamepad_teleop.py", line 399, in mainloop
    self.do_motion()
  File "body/stretch_body/gamepad_teleop.py", line 227, in do_motion
    self.command_robot_joints(robot)
  File "body/stretch_body/gamepad_teleop.py", line 154, in command_robot_joints
    self.wrist_pitch_command.stop_motion(robot)
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/gamepad_joints.py", line 490, in stop_motion
    motor.set_velocity(0,self.params['motion']['max']['accel'])
  File "/home/hello-robot/.local/lib/python3.8/site-packages/stretch_body/dynamixel_hello_XL430.py", line 551, in set_velocity
    if self.params['set_safe_velocity'] and self.in_vel_brake_zone: # only when sentry is active
KeyError: 'set_safe_velocity'`

Could you please help me how to fix this?

Thanks,
Minyoung

hello-binit, 2024-01-25:
Hello @Minyoung1005, thank you for opening this Github issue. The `KeyError: 'vel_brakezone_factor'` and `KeyError: 'set_safe_velocity'` exceptions you're seeing indicate the library is missing keys. I suspect the reason is because the library is out-of-date. I believe if you update your Stretch Body library using `pip3 install -U hello-robot-stretch-body`, and try running the script again, those exceptions should disappear. Could you try this and let me know if it resolves the issue for you?

Minyoung1005, 2024-01-26:
I updated the hello-robot-stretch-body package and it works now!
Also, speed control with the trigger buttons is very easy and intuitive.
Thank you for helping me with this issue :)

closed 2024-01-26 by Minyoung1005

hello-binit, 2024-01-26:
Glad to hear it @Minyoung1005! :)