hello-robot/stretch_body #344: Lift falls with heavier tool before homing

hello-lamsey, 2024-07-25:
[Stretch 3] With the tablet tool attached, the lift float does not hold position after booting up until after the robot is homed. It seems like the calibrated value for lift feedforward current might not be loaded until robot is homed.

Steps to recreate:
* `stretch_body` v0.7.24
* Attach tablet tool (or another tool that's heavier than the gripper)
* Run `REx_calibrate_gravity_comp.py` to set new gravity compensation values
* Shut off and cold boot the robot
* Before homing, lift will not hold up tablet

hello-binit, 2024-07-25:
Thanks @hello-lamsey, would you run `stretch_params.py | grep feedforward` and post the output here?

hello-lamsey, 2024-07-25:
```bash
hello-robot@stretch-se3-3043:~$ stretch_params.py | grep feedforward
stretch_body.robot_params.nominal_params                               param.arm.i_feedforward                                                0                             
stretch_body.robot_params.nominal_params                               param.hello-motor-arm.gains.i_safety_feedforward                       0.0                           
stretch_body.robot_params.nominal_params                               param.hello-motor-left-wheel.gains.i_safety_feedforward                0.0                           
stretch_configuration_params.yaml                                      param.hello-motor-lift.gains.i_safety_feedforward                      1.8                           
stretch_body.robot_params.nominal_params                               param.hello-motor-right-wheel.gains.i_safety_feedforward               0.0                           
stretch_configuration_params.yaml                                      param.lift.i_feedforward                                               1.8   
```