hello-robot/stretch_body #365: Data Classes to replace status and parameter dictionaries, to reduce magic strings and improve readability and api usage.

hello-robot-shehab, 2025-03-03:
For new developers of `stretch_body`, it is quite difficult to find and remember the dictionary key names used for device statuses and Stretch running parameters. 

This is a proposal to convert the dictionaries into [data classes](https://docs.python.org/3.10/library/dataclasses.html).

This change will allow intellisense to help developers quickly access statuses and parameters, and reduce magic strings in the codebase.

Here is an example of a dataclass definition for StepperBase::status:
```python
self.status = {'mode': 0, 'effort_ticks': 0, 'effort_pct':0,'current':0,'pos': 0, 'vel': 0, 'err':0,'diag': 0,'timestamp': 0, 'debug':0,'guarded_event':0,
                       'transport': self.transport.status,'pos_calibrated':0,'runstop_on':0,'near_pos_setpoint':0,'near_vel_setpoint':0,
                       'is_moving':0,'is_moving_filtered':0,'at_current_limit':0,'is_mg_accelerating':0,'is_mg_moving':0,'calibration_rcvd': 0,'in_guarded_event':0,
                       'in_safety_event':0,'waiting_on_sync':0,'in_sync_mode':0,'trace_on':0,'ctrl_cycle_cnt':0,
                       'waypoint_traj':{'state':'idle','setpoint':None, 'segment_id':0,},
                       'voltage':0}
```

Which would become:

```python
from dataclasses import dataclass

class WaypointTrajectoryStatus(dataclass):
        state:str
        setpoint:list[float,float,float]|None
        segment_id: int

class StepperBaseStatuses(dataclass):
    mode: int
    effort_ticks: int
    effort_pct: int
    current: int
    pos: int
    vel: int
    err: int
    diag: int
    timestamp: int
    debug: int
    guarded_event: int
    transport: int
    pos_calibrated: int
    runstop_on: int
    near_pos_setpoint: int
    near_vel_setpoint: int
    is_moving: int
    is_moving_filtered: int
    at_current_limit: int
    is_mg_accelerating: int
    is_mg_moving: int
    calibration_rcvd: int
    in_guarded_event: int
    in_safety_event: int
    waiting_on_sync: int
    in_sync_mode: int
    trace_on: int
    ctrl_cycle_cnt: int
    waypoint_traj: WaypointTrajectoryStatus
    voltage: int

    # For backwards compatibility, allow users to access this with square brackets:
    def __getitem__(self, key):
        return getattr(self, key)

    @staticmethod
    def fromJson(json:dict):
         for k, v in dictionary.items():
             setattr(self, k, v)
```

hello-binit, 2025-03-03:
I like this proposal! Why subclass `dataclass` instead of using the decorator?

hello-robot-shehab, 2025-03-03:
> Why subclass dataclass instead of using the decorator?
Decorator is good! Typo on my part. Thanks!

hello-binit, 2025-03-03:
Cool! Can you make a list of dictionaries this change would include? I'd like to assess how big a change we're talking.

hello-robot-shehab, 2025-03-25:
Hi @hello-binit, I made dataclasses for all the statuses here: https://github.com/hello-robot/stretch_body/tree/feature/enums-and-dataclasses-for-statuses/body/stretch_body/models Thanks!